jasongin/noble-uwp #1: ERror building

farfromrefug, 2017-03-08:
Hi,
I am trying your module.
I have win 10 Insider 10.0.15042.0 with the SDK.
Though it won't build:
```
..\_nodert_generated.cpp(64): error C2039: 'BluetoothAdapter': is not a member of 'Windows::Devices::Bluetooth' [C:\Users\mguillon\dev\aryballe\noble-uwp\uwp\windows.devices .bluetooth\build\binding.vcxproj]
```
I guess this is because it actually builds with VC14 and not VC17.
How did you get node-gyp to use VC17?

BTW i forked your repo and already refactored the gyp files to prevent duplicate and to make it easier to update the win SDK version. I can make a PR if you want

vohtaski, 2017-03-16:
imho, PR would be cool, so that windows build has to be changed in one place :)

farfromrefug, 2017-03-16:
PR created. @jasongin could you help me with the build error?
Thanks

jasongin, 2017-03-16:
Sorry I must have missed the notification when this issue was first opened.

VC17 or updated node-gyp should not be required to build. From the error above, it looks like it is not finding the right version of WinRT metadata. Could it be related to the refactoring you did?

I will try to investigate more later today.

jasongin, 2017-03-17:
Oh, you do have to specify `--msvs_version=2015` with `node-gyp`. And that requires VC 2015.

Did you try using the [`build.cmd`](https://github.com/jasongin/noble-uwp/blob/master/build.cmd) script? Or `npm install` which calls it? That should build everything correctly if you have the necessary prereqs installed.

I checked and there have been no changes to WinRT APIs in the imported namespaces between SDK versions 10.0.15021 (the version currently referenced in the sources) and version 10.0.15052 (the [latest available version](https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewSDK)). So any version within that range should work by just updating the version references.

jasongin, 2017-03-17:
BTW, if you start working with this project / hacking on it, note there is a [develop](https://github.com/jasongin/noble-uwp/tree/develop) branch that has a few additional changes, that are not yet tested as well.

I haven't had much time to work on this project recently but I hope to get back to it soon.

farfromrefug, 2017-03-17:
@jasongin Yes did not see that! Do you want me to cose that PR and open it on develop?

jasongin, 2017-03-20:
That PR is fine against master, since it just touches .gyp files, and the only changes in develop are in the JS code, so there are no conflicts.

farfromrefug, 2017-03-22:
@jasongin i switched to your develop branch to make sure we are using the same code.
Now it still does not build, same error.
I even switched to 15052 too.
Now something i find weird is that i don't have a ```Windows.Devices.Bluetooth``` folder in my sdk ```References``` folder.
Do you have one?

jasongin, 2017-03-22:
Are you using `build.cmd` or `npm install` to build?

All the WinRT types are referenced via `%ProgramFiles(x86)%/Windows Kits/10/UnionMetadata/%WIN_SDK_VER%\Windows.winmd`. (You can open that file in VS Object Browser to see what types are in it.) Each `binding.gyp` file references that Windows.winmd via `AdditionalUsingDirectories`.

The `develop` and `master` branches are identical at the moment so it doesn't matter which you use.

farfromrefug, 2017-03-24:
@jasongin i was using ```npm install```. I tried ```build.cmd``` and still no luck.
BTW i only have C++ build tools, no VS IDE. So i can't try the object browser.
Do you think it might be an issue?
I reall don't see what's going on. Still trying to figure it out...

jasongin, 2017-03-24:
I don't think you should need the VS IDE, as long as you have the C++ build tools and the Windows SDK. But I haven't tried that configuration.

Can you collect a verbose msbuild log and put it on a gist? Then maybe I can figure out what's wrong. Something like these commands should do it:

```cmd
cd build\uwp\windows.devices.bluetooth
"%ProgramFiles(x86)\MSBuild\14.0\Bin\MSBuild.exe" binding.vcxproj /p:Configuration=Release /p:Platform=x64 /v:diag > build.log
```

Note the `binding.vcxproj` is generated by `node-gyp`, so you must have at least attempted building that way first. If you get an error about `nan.h` not found, you might have to do a `npm install` in the `uwp\windows.devices.bluetooth` directory first.

qdot, 2017-04-08:
I'm having the same problem. Using VS2015 with 15063 SDK. Updated common.gypi file, but still getting build errors

Here's a gist of stderr/stdout logs when running build.cmd: https://gist.github.com/qdot/eb1c729b04f82f069899115f92ed6555

jasongin, 2017-04-08:
@qdot The reason I asked for a verbose MSBuild log above is that would hopefully show exactly which .winmd files are referenced during the build and how. The regular node-gyp output doesn't include that information, so I can't see what is wrong from that.

qdot, 2017-04-08:
Ah, sorry about that. Here's the requested log using the msbuild command line specified above.

https://gist.github.com/qdot/a8c2d294c8c6cd546082b239d18005dc

jasongin, 2017-04-08:
Thanks. I'm not sure, but I think the problem might be because the VC `AdditionalUsingDirectories` also includes `%ProgramFiles(x86)%\Windows Kits\10\UnionMetadata` (without the version subdirectory), and that directory also contains a `Windows.winmd` file; I don't know what version SDK version that one is or is supposed to be, but if it doesn't have the required types then it could be the problem.

I just pushed a change https://github.com/jasongin/noble-uwp/commit/2d9a7b0a9bef8b1091855c107142fbfe1acf7f9a to the develop branch that further simplifies the .gyp files and eliminates that unnecessary directory reference. Can you see if that fixes the build for you?

qdot, 2017-04-08:
Pulled changes, moved to develop branch, and changed SDK version to 15063 in the common.gypi, and everything builds now! Thanks!

jasongin, 2017-04-08:
Yeah, I should update the version to 15063, since that is (I think) the release build of the Creators Update SDK.

qdot, 2017-04-08:
Yup, it is. You can get the creators update and the SDK manually at the moment, without being on the insider program now. That's how I got both of them, just wanted to get a bit of a jump on things before Tuesday. :)

farfromrefug, 2017-04-10:
@jasongin amazing! Got it to work just fine!
Now i got it integrated in my app and it seems to be working almost fine.
I have an issue with the discovery.
In my log i see this
```
Mon, 10 Apr 2017 14:08:50 GMT noble-uwp watcher received: 00:1b:dc:06:cb:ed NeOse Dev Martin
Mon, 10 Apr 2017 14:08:50 GMT noble-uwp     address type: public
Mon, 10 Apr 2017 14:08:50 GMT noble-uwp     advertisement type: scanResponse
Mon, 10 Apr 2017 14:08:50 GMT noble-uwp     data section: shortenedLocalName
Mon, 10 Apr 2017 14:08:50 GMT noble-uwp     data section: manufacturerSpecificData
Mon, 10 Apr 2017 14:08:50 GMT noble-uwp     flags: undefined
DEBUG neose:handler:bluetooth discover { id: '001bdc06cbed',
  type: 'bluetooth',
  state: 'disconnected',
  address: '00:1b:dc:06:cb:ed' }
```
noble seems to discover my device named "NeOse Dev Martin" perfectly. However the emitted event does not contain all necessary data as we can have in noble like detailed [here](https://github.com/sandeepmistry/noble#peripheral-discovered)
Is that something that is not implemented yet?

Thanks

farfromrefug, 2017-04-10:
@jasongin amazing! builds perfectly now.
I already integrated it in my app.
Now i am getting into it. Will create different issues.

closed 2017-04-10 by farfromrefug