jasongin/noble-uwp #5: discoverCharacteristics failing when a valid characteristic UUID is passed

qdot, 2017-04-09:
Trying to run a script that works on noble on OS X/linux, that uses the discoverCharacteristics([uuids, ...], callback()) format. The uuids are formatted without -'s and using lower case for hex, while it looks like windows is returning uuids with -'s and upper case hex. Should probably clean up uuid format in the filteruuid function before comparison?

qdot, 2017-04-09:
BTW, patch incoming, just putting this here in case I get interrupted and don't get it in soon. :)

qdot, 2017-04-09:
Ok, revising initial report. There's already a uuid normalization function I didn't see, looks like the problem is that the characteristic object is passed to filterUuid, instead of just the uuid. So discoverCharacteristic will always fail unless no uuid is passed.

jasongin, 2017-04-10:
Fixed by https://github.com/jasongin/noble-uwp/commit/27fa5e8542cfecdd4fe7e48df68ad437caf525ee

PR #6 (2017-04-10): Fix characteristic filtering
Characteristic objects were being sent to filterUuid, instead of their normalized UUID. This patch fixes that so that characteristic filtering works again.

Fixes #5
commit 27fa5e85 (2017-04-10): Fix characteristic filtering (#6)

Characteristic objects were being sent to filterUuid, instead of their normalized UUID. This patch fixes that so that characteristic filtering works again.

Fixes #5
closed 2017-04-10 by jasongin