jasongin/noble-uwp #7: advertisment always empty

farfromrefug, 2017-04-10:
I now got it work and it is already discovering a lot of devices !
Now about the device that i am interested in. The advertisment data is always empty
```
{ localName: null,
  txPowerLevel: null,
  manufacturerData: null,
  serviceUuids: [],
  serviceData: [] }
```
I see this in my log before i get the ```'discover'``` event:
```
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp watcher received: 5c:7e:e3:ca:4e:09 
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     address type: public
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     advertisement type: connectableUndirected
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     data section: flags
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     data section: manufacturerSpecificData
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     flags: 6
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp watcher received: 5c:7e:e3:ca:4e:09 
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     address type: public
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     advertisement type: scanResponse
Mon, 10 Apr 2017 14:28:13 GMT noble-uwp     flags: undefined
```
Maybe advertisment is not already implemented?

farfromrefug, 2017-04-10:
I now see in the code that manufacturerData is not implemented yet.
Will try to implement it and make a PR

farfromrefug, 2017-04-10:
@jasongin i am new to nodeRT dev.
Do you know how i can convert a ```IBuffer``` to a node ```Buffer```?
I am trying to get the ```data``` from [BluetoothLEManufacturerData](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothlemanufacturerdata)

jasongin, 2017-04-10:
Cool, a PR would be appreciated.

Use [`rt.toBuffer()` from `rt-utils.js`](https://github.com/jasongin/noble-uwp/blob/master/lib/rt-utils.js#L70) to convert from a WinRT `IBuffer` to a Node `Buffer`.

farfromrefug, 2017-04-11:
@jasongin ok ```rt.toBuffer()``` works. Now the issue is that ```IBuffer``` does not seem to contain what it should.
I am investigating

farfromrefug, 2017-04-18:
created the PR https://github.com/jasongin/noble-uwp/pull/9

jrobeson, 2017-05-18:
#9 was merged, so this can be closed?

farfromrefug, 2017-05-18:
Yes it's done !

closed 2017-05-18 by farfromrefug

jasongin, 2017-05-18:
It was merged into the `develop` branch. I still need to merge into the `master` branch and publish a new package to NPM. I'll do that soon!

jasongin, 2017-05-18:
Merged to `master` and published to NPM as v0.3.1.

hoffmannjan, 2017-06-03:
Hi, as I see advertisement now exsists, but on all my Apple Devices (I don't have others at the moment to test It) I see localName: null.

Have this problem too?

jasongin, 2017-06-05:
@hoffmannjan Not all devices supply a localName in their advertisement. Do you see a localName value when running the same code using noble on a non-Windows device?

hoffmannjan, 2017-06-06:
Yes, when I'm using noble on MacOS detected deveices have theier local names like 'Jan Hoffmanns iPhone'.
When I switch to ASUS laptop, all of them has empty local name.

Will try to test with another BT stick in Asus.