jasongin/noble-uwp #8: crash on node 7.9.0

farfromrefug, 2017-04-18:
With the latest nodejs, ```noble-uwp``` crashes every time
It works with the same test on  6.x.
Could you update the uwp projects for node 7.x ? 
I think it might solve the issue.
Thanks

```
noble
on -> stateChange: poweredOn
on -> discover:  undefined null
on -> discover:  undefined NeOse Pro Backup 1.3.35
on -> discover:  undefined NeOse Dev Martin
Stacktrace (bbbbbbbb-bbbbbbbb) 00000109134038B9 0000000000000000:
==== JS stack trace =========================================

Security context: 00000109134266A1 <JS Object>#0#
    1: checkCommunicationResult(aka checkCommunicationResult) [C:\dev\noble-uwp\lib\bindings.js:746] [pc=000000106CEBA220](this=0000024244802311 <undefined>,deviceUuid=000002894B1AACB1 <String[12]: 001bdc06cbed>,result=00000109134038B9 <FixedArray[189]>#1#)
    2: /* anonymous */(aka /* anonymous */) [C:\dev\noble-uwp\lib\bindings.js:170] [pc=000000106CEB9F01](this=0000024244802311 <undefined>,result=00000109134038B9 <FixedArray[189]>#1#)
    3: PromiseHandle(aka PromiseHandle) [native promise.js:1] [pc=000000106CEA8B42](this=0000024244802311 <undefined>,F=00000109134038B9 <FixedArray[189]>#1#,J=000002894B1B1309 <JS Function rt.promisify.then.result (SharedFunctionInfo 000001204D1957B9)>#2#,K=000002894B1B14F9 <an Object with map 0000001ECCAAE1A9>#3#)
    4: /* anonymous */(aka /* anonymous */) [native promise.js:1] [pc=000000106CEA87EA](this=0000024244802311 <undefined>)

==== Details ================================================

[1]: checkCommunicationResult(aka checkCommunicationResult) [C:\dev\noble-uwp\lib\bindings.js:746] [pc=000000106CEBA220](this=0000024244802311 <undefined>,deviceUuid=000002894B1AACB1 <String[12]: 001bdc06cbed>,result=00000109134038B9 <FixedArray[189]>#1#) {
  // expression stack (top to bottom)
  [03] : 000001204D196271 <FixedArray[20]>#4#
  [02] : 2
  [01] : 0000010913403501 <String[6]: status>
  [00] : 00000109134038B9 <FixedArray[189]>#1#
--------- s o u r c e   c o d e ---------
function checkCommunicationResult(deviceUuid, result) {\x0d\x0a\x09if (result.status === GattCommunicationStatus.unreachable) {\x0d\x0a\x09\x09throw new Error('Device unreachable: ' + deviceUuid);\x0d\x0a\x09} else if (result.status === GattCommunicationStatus.protocolError) {\x0d\x0a\x09\x09throw new Error('Protocol error communicating with device: ' + deviceUuid);\x0d\x0a\x09}\x0d\x0a}
-----------------------------------------
}

[2]: /* anonymous */(aka /* anonymous */) [C:\dev\noble-uwp\lib\bindings.js:170] [pc=000000106CEB9F01](this=0000024244802311 <undefined>,result=00000109134038B9 <FixedArray[189]>#1#) {
  // stack-allocated locals
  var services = 0000024244802351 <the hole>
  var serviceUuids = 0000024244802351 <the hole>
  // expression stack (top to bottom)
  [05] : 00000109134038B9 <FixedArray[189]>#1#
  [04] : 000002894B1AACB1 <String[12]: 001bdc06cbed>
  [03] : 0000024244802311 <undefined>
  [02] : 0000026E92F233D9 <JS Function checkCommunicationResult (SharedFunctionInfo 000001998EE5D299)>#5#
--------- s o u r c e   c o d e ---------
function result => {\x0d\x0a\x09\x09checkCommunicationResult(deviceUuid, result);\x0d\x0a\x0d\x0a\x09\x09let services = rt.toArray(result.services);\x0d\x0a\x09\x09let serviceUuids = services.map(s => formatUuid(s.uuid))\x0d\x0a\x09\x09\x09.filter(filterUuids(filterServiceUuids));\x0d\x0a\x0d\x0a\x09\x09debug(deviceUuid + ' services: %o', serviceUuids);\x0d\x0a\x09\x09this.emit('servicesDiscov...

-----------------------------------------
}

[3]: PromiseHandle(aka PromiseHandle) [native promise.js:1] [pc=000000106CEA8B42](this=0000024244802311 <undefined>,F=00000109134038B9 <FixedArray[189]>#1#,J=000002894B1B1309 <JS Function rt.promisify.then.result (SharedFunctionInfo 000001204D1957B9)>#2#,K=000002894B1B14F9 <an Object with map 0000001ECCAAE1A9>#3#) {
  // stack-allocated locals
  var D = 0000024244802421 <false>
  var L = 0000024244802311 <undefined>
  // expression stack (top to bottom)
  [06] : 00000109134038B9 <FixedArray[189]>#1#
  [05] : 0000024244802311 <undefined>
  [04] : 000002894B1B1309 <JS Function rt.promisify.then.result (SharedFunctionInfo 000001204D1957B9)>#2#
  [03] : 000001091341ADE1 <FixedArray[45]>#6#
  [02] : 000001091341ADE1 <FixedArray[45]>#6#
--------- s o u r c e   c o d e ---------
function PromiseHandle(F,J,K){\x0avar D=(%_DebugIsActive()!=0);\x0atry{\x0aif(D)%DebugPushPromise(K.promise);\x0avar L=J(F);\x0aK.resolve(L);\x0a}%catch(exception){\x0atry{K.reject(exception);}catch(e){}\x0a}finally{\x0aif(D)%DebugPopPromise();\x0a}\x0a}
-----------------------------------------
}

[4]: /* anonymous */(aka /* anonymous */) [native promise.js:1] [pc=000000106CEA87EA](this=0000024244802311 <undefined>) {
  // stack-allocated locals
  var P = 0000024244802311 <undefined>
  // expression stack (top to bottom)
  [05] : 000002894B1B14F9 <an Object with map 0000001ECCAAE1A9>#3#
  [04] : 000002894B1B1309 <JS Function rt.promisify.then.result (SharedFunctionInfo 000001204D1957B9)>#2#
[... 77 lines cut ...]
                 3: 2326624
                 4: 234921986
 #18# 00000242448049F9: 00000242448049F9 <Symbol: uninitialized_symbol>
 #19# 0000024244802559: 0000024244802559 WeakCell for 0
 #20# 000001091341AF59: 000001091341AF59 <JS Function (SharedFunctionInfo 0000024244821B49)>
 #21# 0000010913406891: 0000010913406891 <JS Function InternalArray (SharedFunctionInfo 0000024244848981)>
 #22# 0000024244804429: 0000024244804429 <Symbol: promise_async_stack_id_symbol>
 #23# 00000242448045F1: 00000242448045F1 <Symbol: promise_handled_by_symbol>
 #24# 0000024244804539: 0000024244804539 <Symbol: promise_forwarding_handler_symbol>
 #25# 00000242448046A1: 00000242448046A1 <Symbol: promise_has_handler_symbol>
 #26# 0000024244804749: 0000024244804749 <Symbol: promise_reject_reactions_symbol>
 #27# 000001204D196159: 000001204D196159 <FixedArray[11]>
                 0: 1
                 1: 0
                 2: 4
                 3: 7
                 4: 0000024244802D19 <String[5]: Error>
                 5: 0000024244802311 <undefined>
                 6: 0000024244802311 <undefined>
                 7: 0000024244802311 <undefined>
                 8: 0000024244802311 <undefined>
                 9: 0000024244802311 <undefined>
                  ...
=====================
```

jasongin, 2017-04-25:
@farfromrefug Did you rebuild noble-uwp after switching from node v6 to node v7? Node addons are not binary-compatible across node major versions. (I'm actually part of a [working group that is fixing that problem](https://github.com/nodejs/abi-stable-node), but it's a longer-term solution.)

jasongin, 2017-04-25:
If that's not the problem, then this would likely be a NodeRT issue. It might be one more reason to [not use NodeRT in this project](https://github.com/jasongin/noble-uwp/issues/4).

farfromrefug, 2017-04-25:
@jasongin Yes that issue appears while noble-uwp is actually built with node v7 by doing 
```npm rebuild --runtime-electron --target=1.6.6 --arch=x64 --rebuild```

But when did you actually create the project? (cpp files). It seems that node rt made some [fixes](https://github.com/NodeRT/NodeRT/pull/41) for Electron 
And it seems like you use files generated with nodeRT. maybe they are not up to date?

jasongin, 2017-04-25:
I last generated the files with the latest NodeRT maybe a month or two ago, so it was long after those linked electron fixes.

I'll try to investigate with Node version 7, and open bugs against NodeRT depending on what I find.

farfromrefug, 2017-04-25:
@jasongin i can do it too if you don't have time ;) Is there s like an easy command to generate the sources from nodert?
Thanks

jasongin, 2017-04-25:
I probably won't have time to work on this for at least a few days, so go for it!

The NodeRT code-generation commands are saved at https://github.com/jasongin/noble-uwp/blob/master/uwp/generate-uwp.cmd

Note I had to manually patch the generated code [in one place](https://github.com/jasongin/noble-uwp/blob/master/uwp/windows.devices.bluetooth.genericattributeprofile/_nodert_generated.cpp#L13660) to fix a build error. I still need to open a bug against NodeRT for that...

I think re-generating the code is unlikely to fix anything, since I don't see any recent updates to NodeRT.

farfromrefug, 2017-04-26:
@jasongin thanks will try. Even if it does not fix anything at least i can report saying we tried with the latest build!

nadavbar, 2017-05-12:
Hi @jasongin and @farfromrefug , I believe that the issue here was due to a change in the memory management policy in the v8 versions that node 7.* picks up and it did cause a dormant bug in NodeRT to appear. I just fixed it in the latest NodeRT release (2.0.5), and I have a good reason to believe that it should fix this issue as well.

I didn't have the chance to reproduce this exact issue, but in case you guys can verify that this issue is now fixed, that would be super helpful :)

jasongin, 2017-05-12:
Thanks @nadavbar! @farfromrefug or I will verify that this is fixed.

farfromrefug, 2017-05-13:
Thanks @nadavbar !
@jasongin do i just need to build or do we need to re generate the project?

nadavbar, 2017-05-16:
@farfromrefug  - Another option is just to replace the NodeRTUtils.cpp file in all of the modules (each NodeRT module has it's own version) and then rebuild. You can grab it from here: https://github.com/NodeRT/NodeRT/blob/master/src/NodeRTLib/ProjectFiles/NodeRtUtils.cpp

farfromrefug, 2017-05-16:
@nadavbar that's a lot easier ;) Thanks
Will try today then

farfromrefug, 2017-05-16:
@jasongin @nadavbar It works! I dont see anymore crash using noble-uwp
Thanks a lot!

jasongin, 2017-05-19:
I merged the fix here:https://github.com/jasongin/noble-uwp/commit/2ba36a460b6632afa251dc30fbd75eb804e6ed8d

closed 2017-05-19 by jasongin

farfromrefug, 2017-05-19:
@jasongin thanks! will close my PR then