jasongin/noble-uwp #10: error C2011: 'timespec': 'struct'

FabianKLS, 2017-04-24:
When installing Noble-UWP I get a couple of the same errors: error C2011: 'timespec': 'struct'. Not sure how to solve this.

jasongin, 2017-05-16:
@FabianKLS Can you share more details? What line of code is the error referring to?

Sorry I must have missed the notification when this issue was originally reported.

hoffmannjan, 2017-06-02:
```
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  core.c
  descriptor.c
  hotplug.c
  io.c
C:\Users\janhoffmann\Desktop\node-sensortag\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'time
spec': 'struct' type redefinition (compiling source file ..\libusb\libusb\descriptor.c)C:\Users\janhoffmann\Desktop\nod
e-sensortag\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinitio
n (compiling source file ..\libusb\libusb\io.c) [C:\Users\janhoffmann\Desktop\node-sensortag\node_modules\usb\build\lib
usb.vcxproj]
C:\Users\janhoffmann\Desktop\node-sensortag\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'time
spec': 'struct' type redefinition (compiling source file ..\libusb\libusb\hotplug.c) [C:\Users\janhoffmann\Desktop\node
-sensortag\node_modules\usb\build\libusb.vcxproj]

C:\Users\janhoffmann\Desktop\node-sensortag\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'time
spec': 'struct' type redefinition (compiling source file ..\libusb\libusb\core.c)C:\Program Files (x86)\Windows Kits\10
\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\desc
riptor.c)C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespe
c' (compiling source file ..\libusb\libusb\io.c) [C:\Users\janhoffmann\Desktop\node-sensortag\node_modules\usb\build\li
busb.vcxproj]
```

in my case this one

jasongin, 2017-06-02:
> node-sensortag\node_modules\usb

This is an issue in the `usb` package. You should report it over there: https://github.com/tessel/node-usb

Actually it looks like there are a few open issues already there that reference the "timespec" error:
https://github.com/tessel/node-usb/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20timespec

jasongin, 2017-06-27:
Closing this since it's an external issue already tracked by the `node-usb` project.

closed 2017-06-27 by jasongin