jasongin/noble-uwp #11: errors when trying npm install noble-uwp

Janneman84, 2017-05-01:
Sorry for the big dump below, but I'm trying to install noble-uwp and it gives me a bunch of errors. It looks like there is a problem with the Bluetooth. May it have something to do with me running the latest version of Windows (15063) instead of the preview version? Also I'm running the latest NodeJS on my system.

I noticed it builds all the NodeRT stuff, while it only needs the Bluetooth ones right?

I'm planning on using it in my Electron application. Any help is greatly appreciated!

C:\Projects\Diagnoptics>npm install noble-uwp

> usb@1.2.0 install C:\Projects\Diagnoptics\node_modules\usb
> node-pre-gyp install --fallback-to-build

[usb] Success: "C:\Projects\Diagnoptics\node_modules\usb\src\binding\usb_bindings.node" is installed via remote

> bluetooth-hci-socket@0.5.1 install C:\Projects\Diagnoptics\node_modules\bluetooth-hci-socket
> node-gyp rebuild

C:\Projects\Diagnoptics\node_modules\bluetooth-hci-socket>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  win_delay_load_hook.cc
  Generating code
  Finished generating code
  binding.vcxproj -> C:\Projects\Diagnoptics\node_modules\bluetooth-hci-socket\build\Release\\binding.node

> noble-uwp@0.2.0 install C:\Projects\Diagnoptics\node_modules\noble-uwp
> build.cmd

Building NodeRT UWP adapter for Windows.Foundation
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
  CollectionsConverterUtils.cpp
  win_delay_load_hook.cc
     Creating library C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\windows.foundation\build\Release\binding.lib and object C:\Projects\Di
  agnoptics\node_modules\noble-uwp\uwp\windows.foundation\build\Release\binding.exp
  Generating code
  Finished generating code
  binding.vcxproj -> C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\windows.foundation\build\Release\\binding.node

Building NodeRT UWP adapter for Windows.Storage.Streams
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
  CollectionsConverterUtils.cpp
  win_delay_load_hook.cc
     Creating library C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\windows.storage.streams\build\Release\binding.lib and object C:\Projec
  ts\Diagnoptics\node_modules\noble-uwp\uwp\windows.storage.streams\build\Release\binding.exp
  Generating code
  Finished generating code
  binding.vcxproj -> C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\windows.storage.streams\build\Release\\binding.node

Building NodeRT UWP adapter for Windows.Devices.Bluetooth
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
..\_nodert_generated.cpp(64): error C2039: 'BluetoothAdapter': is not a member of 'Windows::Devices::Bluetooth' [C:\Projects\Diagnoptics\node_m
odules\noble-uwp\uwp\windows.devices.bluetooth\build\binding.vcxproj]
..\_nodert_generated.cpp(64): error C2065: 'BluetoothAdapter': undeclared identifier [C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\window
s.devices.bluetooth\build\binding.vcxproj]
..\_nodert_generated.cpp(64): error C2065: 'wintRtInstance': undeclared identifier [C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\windows.
devices.bluetooth\build\binding.vcxproj]
..\_nodert_generated.cpp(65): error C2039: 'BluetoothAdapter': is not a member of 'Windows::Devices::Bluetooth' [C:\Projects\Diagnoptics\node_m
odules\noble-uwp\uwp\windows.devices.bluetooth\build\binding.vcxproj]
..\_nodert_generated.cpp(65): error C2143: syntax error: missing ';' before '^' [C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\windows.dev
ices.bluetooth\build\binding.vcxproj]
..\_nodert_generated.cpp(65): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int [C:\Projects\Diagnoptic
s\node_modules\noble-uwp\uwp\windows.devices.bluetooth\build\binding.vcxproj]
..\_nodert_generated.cpp(65): error C3699: '^': cannot use this indirection on type 'int' [C:\Projects\Diagnoptics\node_modules\noble-uwp\uwp\w
[... 291 lines cut ...]
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules\noble-uwp\node_modules\noble\node_modules\xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "noble-uwp"
npm ERR! node v6.10.2
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! noble-uwp@0.2.0 install: `build.cmd`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the noble-uwp@0.2.0 install script 'build.cmd'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the noble-uwp package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     build.cmd
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs noble-uwp
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls noble-uwp
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Projects\Diagnoptics\npm-debug.log

Janneman84, 2017-05-02:
Oh I also tried installing Visual Studio 2017, including the 15063 SDK. This updated the files in the Windows Kits\10 folder, but this didn't fix the errors.

I'm sure it's just some version issue.

jasongin, 2017-05-15:
Sorry I must have missed the notification when this was posted a couple weeks ago. The errors like this one:

> C2039: 'BluetoothAdapter': is not a member of 'Windows::Devices::Bluetooth' 

seem to indicate some problem related to referencing the `.winmd` files from the correct version of the Windows SDK. The `BluetoothAdapter` class did not exist in older versions of the SDK.

Can you collect a verbose msbuild log and put it on a gist? Then maybe I can figure out what's wrong. (The regular node-gyp output doesn't include details about how the SDK files are referenced.) Something like these commands should do it:

```cmd
cd build\uwp\windows.devices.bluetooth
"%ProgramFiles(x86)\MSBuild\14.0\Bin\MSBuild.exe" binding.vcxproj /p:Configuration=Release /p:Platform=x64 /v:diag > build.log
```

Note the `binding.vcxproj` is generated by `node-gyp`, so you must have at least attempted building that way first.

Janneman84, 2017-05-16:
I tried what you suggested. In the build.log it keeps referring to C:\Program Files (x86)\Windows Kits\10\Include\10.0.10150.0\ and not 10.15063.0.

jasongin, 2017-05-16:
Can you share the log - put it on a gist or something? I might be able to see why it's referring to the wrong version directory.

Janneman84, 2017-05-17:
Okay I put it here:
https://gist.github.com/Janneman84/dd0127a6c778c76aa5af3d802ad44310

In this one I first renamed the 10.0.10150.0 folder to see what would happen. I believe it even tries the Windows 8 kit now...

jasongin, 2017-05-17:
Oh, so it looks like you tried to install from NPM. Unfortunately the version published to NPM was very out of date.

I just published the latest code to NPM. It includes the updated reference to the 10.0.15063 SDK version, as well as other build improvements. Can you try again?

jasongin, 2017-06-09:
This issue appears to be resolved. Closing due to no response.

closed 2017-06-09 by jasongin