jasongin/noble-uwp #12: subscribe does not send notifications

vohtaski, 2017-05-11:
Thank your for this module!

Everything works for me on windows, I can discover and connect the device and send
commands to the device.
However, when I try to subscribe for notification characteristic, I don't get any data.

This code works with noble on Mac, but the callback in `on('data')` is not called on Windows.
Any hints what I might be doing wrong? Thank you in advance!
```js
let readChar = device.characteristics[readCharacteristicUUID];
readChar.on('data', (data, isNotification) => {
  console.log('data');
  console.log(data);
});
readChar.subscribe((err) => {
  if (err) console.log(err);
  console.log('Notification started');
});
```

vohtaski, 2017-05-12:
@jasongin do notifications work for you on 15063?
After debugging, I see the listener below is never called on my windows machine:
```js
				listener = ((source, e) => {
					debug('notification: %s %s %s', deviceUuid, serviceUuid, characteristicUuid);
					let data = rt.toBuffer(e.characteristicValue);
					this.emit('read', deviceUuid, serviceUuid, characteristicUuid, data, true);
				}).bind(this);

				characteristic.addListener('valueChanged', listener);
				this._listenerMap[listenerKey] = listener;
				rt.keepAlive(true);
				this.emit('notify', deviceUuid, serviceUuid, characteristicUuid, notify);
```
https://github.com/jasongin/noble-uwp/blob/develop/lib/bindings.js#L296

jasongin, 2017-05-15:
@vohtaski I don't see anything obviously wrong there. Notifications were working for me on 15063. I've only tested with a TI SensorTag, and not a wide variety of BT devices.

vohtaski, 2017-05-15:
Strange. Is it possible to see the config for SensorTag?
My device sends many notifications for testing,
maybe windows is blocking it ...

farfromrefug, 2017-05-16:
@jasongin I am now facing the same issue. I also debugged the code and the listener is never called.
How does ```characteristic.addListener('valueChanged', listener);``` works? How actually send the ```valueChanged``` event?

farfromrefug, 2017-05-16:
ACctually it seems to be a bug in 15063 with console apps
https://social.msdn.microsoft.com/Forums/en-US/58da3fdb-a0e1-4161-8af3-778b6839f4e1/bluetooth-bluetoothledevicefromidasync-does-not-complete-on-10015063?forum=wdk#ef927009-676c-47bb-8201-8a80d2323a7f

There is a way to resolve it by (comment by Ilis on May 9th) but i don't see how to do this with NodeRT :s

vohtaski, 2017-05-16:
@farfromrefug 
I very dirtily patched the generated _nodert_generated.cpp file
with instructions in your link and it's working for me now.

Still, I guest the best is to wait for proper windows fix... 
Will send a link with patched file soon

vohtaski, 2017-05-16:
Here is what works for me:
https://github.com/vohtaski/noble-uwp/commit/108df4471e3eaedb0a8e4a785f2ca867a9c195fb

farfromrefug, 2017-05-16:
@vohtaski great to know it s working. However i don't think it s a good idea to path the generated nodeRT file, or we will need to patch it everytime we update the NodeRT files...
Could be run that code in the ```binding.js``` using NodeRT?

vohtaski, 2017-05-16:
@farfromrefug
there is anyways one nodeRT file that is already patched :)
Anyways, I guess it's a temporary fix, hopefully before the windows bug is fixed

jasongin, 2017-05-16:
Hmm, maybe when I saw it working was using the 15063 SDK but a different build of Windows. I'm glad you guys were able to figure it out!

Yes, I already had to patch the generated code [in one place](https://github.com/jasongin/noble-uwp/blob/master/uwp/windows.devices.bluetooth.genericattributeprofile/_nodert_generated.cpp#L13660) to workaround a NodeRT bug. It hasn't been too much of a problem so far. Whenever I regenerate the code, I check the diff and manually revert the patched section of code, as well as some `.gyp` file changes.

farfromrefug, 2017-05-16:
@vohtaski @jasongin it works perfectly. I already made my first release with it! THANKS

jasongin, 2017-05-19:
I merged @vohtaski's fix here: https://github.com/jasongin/noble-uwp/commit/0b30cd02b40ce1ce67c9b6a40fc30645587f25a1

closed 2017-05-19 by jasongin