jasongin/noble-uwp #17: node-gyp rebuild noble-uwp

hoffmannjan, 2017-06-09:
Hi,

I'm trying to do electron-rebuild (as well as node-gyp rebuild which is the part of electron-rebuild) noble-uwp and after installing Windows SDK (matching version) and VS 2015 Express and also windows-build-tools i got this error

https://pastebin.com/mfYu34uQ

Any ideas?

I've never messed with any flags, neither in system settings nor in noble-uwp configuration files.

jasongin, 2017-06-09:
Because this package includes 7 separate native modules, not just one, `node-gyp rebuild` alone doesn't do the right thing. That is why there is a `build.cmd` there.

I don't know how you can integrate that into `electron-rebuild`; I don't have experience with Electron development.

I have thought about separating the 7 UWP modules into npm dependency packages, [so that node-pre-gyp can work](https://github.com/jasongin/noble-uwp/issues/13). I guess that would also solve this problem.

hoffmannjan, 2017-06-11:
Cool , the build.cmd do the trick, thanks for answer ;).

closed 2017-06-12 by hoffmannjan

itome, 2017-08-20:
Hi,
I also try to use noble-uwp in my electron project but I cannot.
The error log is next , by seeing this electron and noble-uwp is not completely matched.
```
```
@hoffmannjan seem to done well , so if you have some tips to run noble-uwp in electron. please tell me.
I already tried install.cmd (build.cmd renamed).

Janneman84, 2017-08-21:
Same problem here. I'm not sure what to do to make it work in Electron. I tried using electron-rebuild but that gives me an error: `Undefined variable module_name in uwp\windows.foundation\binding.gyp while loading dependencies of binding.gyp while trying to load binding.gyp`.

Any help would be greatly appreciated.

itome, 2017-08-21:
module_name is likely to be node-pre-gyp dependences. (https://github.com/nodejs/node-gyp/issues/508)
 And node-pre-gyp not completery supported by electron. (https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md)
But I cannot rebuild noble-uwp by node-gyp.

itome, 2017-08-21:
@Janneman84 I found the way to rebuild noble-uwp for electron.
Please see this issue #31

lilliesAndRoses, 2018-06-27:
These are the steps I followed to use noble-uwp on a windows 10 ble enabled machine.
 
I replaced all my `require('noble') `calls with `require('noble-uwp')`, updated the dependencies in package.json from noble to noble-uwp.
Did 
`npm install`

Did the rebuild step using following
`npm rebuild --runtime=electron --target=1.8.1 --arch=x64 --rebuild --disturl=https://atom.io/download/electron --build_from_source=true`

Then again called
`..\node_modules\.bin\build.cmd`

No errors were thrown by last two steps.
Still when I start my app, I do get an error saying 

Error: Cannot find module 'C:\Users\<my_path>\node_modules\noble-uwp\uwp\windows.foundation\binding\electron-v1.8-win32-x64\binding.node'

revati-kiwi, 2019-05-13:
hi @hoffmannjan 

can you help on how to use build.cmd, i am getting this error on an application that is build with electron-builder.

Thanks