jasongin/noble-uwp #19: allowDuplicates problem

hoffmannjan, 2017-06-23:
Hi, as I see in [source code](https://github.com/jasongin/noble-uwp/blob/master/lib/bindings.js#L74-L81) the option about allowDuplicates isn't interpreted in any way.
Is this because of lack of setting this in Windows 10 Bluetooth API (can't find anything about this in documentation) or I should configure this in other place?

jasongin, 2017-06-27:
Right, the Windows 10 Bluetooth API doesn't provide any way to ignore duplicate scan results. However, I think this could be implemented in `noble-uwp`, in [the `_onAdvertisementWatcherReceived()` callback](https://github.com/jasongin/noble-uwp/blob/master/lib/bindings.js#L435), by keeping track of what addresses have been scanned so far and using that list to avoid emitting duplicate `"discover"` events.

You're welcome to submit a PR! Otherwise I will try to get to this eventually.

jasongin, 2017-07-28:
This is implemented now, in version 0.5.0. Can  you confirm that it works how you expect?

hoffmannjan, 2017-08-07:
Yes, can confirm! Thanks!

closed 2017-08-07 by hoffmannjan