jasongin/noble-uwp #23: startScanning() does not filter by service UUID

jasongin, 2017-07-12:
Reported at https://github.com/sandeepmistry/noble/pull/646#issuecomment-314676439

VinciShark, 2017-07-12:
I will dig into it……just find that native binding never get uuid filter arg

VinciShark, 2017-07-12:
@jasongin Don't know what dataType transfer need to use
just try 
```
BluetoothLEAdvertisement = (function () {
  var cls = function BluetoothLEAdvertisement() {
    this.localName = new String();
    this.flags = new BluetoothLEAdvertisementFlags();
    this.dataSections = new Object();
    this.manufacturerData = new Object();
    this.serviceUuids = [0x0000000000001000800000805f9b34fb];
  };
……
```

but nothing change

nykma, 2017-07-12:
What's the type of `this.serviceUuids`? Seems like `new Objcet()` is not the real type accepted by Windows BLE binding.

jasongin, 2017-07-12:
In [the Windows API](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisement#Windows_Devices_Bluetooth_Advertisement_BluetoothLEAdvertisement_ServiceUuids) it is an `IList<Guid>`. However NodeRT projects GUIDs to JS as strings. So `serviceUuids` should be a list of strings, with the UUIDs formatted as GUID strings: `{00000000-0000-0000-0000-000000000000}`

So I _think_ the fix in `startScanning()` should include something like this:
```JavaScript
serviceUuids.forEach(uuid => {
  var uuidString = // format uuid as GUID string
  this._advertisementWatcher.advertisementFilter.advertisement.serviceUuids.add(uuidString);
});
this._advertisementWatcher.start();
```

jasongin, 2017-07-13:
Note bluetooth uuids can be in short (uint32) and long (guid) form. I recently discovered there is a [`BluetoothUuidHelper`](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothuuidhelper) class that can convert between them. That could be useful here, and probably should have been used in the existing `formatUuid()` function.

VinciShark, 2017-07-22:
give up cause failed to understand code NodeRT generated

jasongin, 2017-07-24:
@VinciShark OK, I'll pick this up.

jasongin, 2017-07-28:
This is implemented now, in version 0.5.0. Can someone confirm that it works how you expect?

VinciShark, 2017-07-28:
@jasongin  Will test it today, just finish holiday

jasongin, 2017-08-26:
This might not be working correctly, according to https://github.com/sandeepmistry/noble/pull/646#issuecomment-324438544