jasongin/noble-uwp #24: .notify() & .subscribe stoped working after 0.4.4 upgrade

hoffmannjan, 2017-07-14:
`charREAD.on('read', function(data, err) {
	console.log('>>> charREAD notify', data, err);
});

charREAD.subscribe()
`

This code stopped working :(.

jasongin, 2017-07-17:
Are you sure? None of the [changes since 0.3.2](https://github.com/jasongin/noble-uwp/compare/2ba36a460b6632afa251dc30fbd75eb804e6ed8d...master) have affected any code related to notify & subscribe.

jasongin, 2017-07-17:
Also, the test I use all the time includes subscribe & notify, and it still works fine.

hoffmannjan, 2017-07-18:
100% sure, tested this again and have the same problem.
I was also astonished looking in git diff of this update, that something like this can stopped working.

hoffmannjan, 2017-07-19:
Here's the code that I use for this test
https://pastebin.com/X9xeiRrR

jasongin, 2017-07-19:
That code looks straightforward. So this still doesn't make sense to me. I'll try to investigate more when I have time.

jasongin, 2017-07-30:
Is anyone else having this problem?

taktran, 2017-07-31:
I've tried this with a stripped down app, and can confirm that the `.notify`'s stopped working from `0.3.2` onwards

I've tried

* `0.4.2` to `0.5.1` - didn't get any callbacks
* `0.4.1` - didn't compile for me
* `0.3.2` - worked! Would trigger callbacks on notify

jasongin, 2017-07-31:
@taktran Thanks for checking all the versions. That narrowed it down, then I found the problem!

One of the places where I had manually patched the generated code got accidentally reverted:
https://github.com/jasongin/noble-uwp/commit/6126c389b8dc38834090b55efc593b3d0e01b49d#diff-f5cdb08420f05042c8348dbd32ee22d6L6603

I originally made that patch because it was crucial to enable notifications. I'm just not sure why notifications continued to work for me after it was reverted, so that I didn't catch that mistake when testing.

Anyway, I'll fix this and publish an update today.

jasongin, 2017-08-01:
Fixed in 0.5.2.

closed 2017-08-01 by jasongin

taktran, 2017-08-01:
Have tested on 0.5.2, and notifications work as expected. Thanks a bunch @jasongin ! 👍