jasongin/noble-uwp #25: "Device is not connectable" error randomly occurs when connecting

taktran, 2017-07-24:
Hi,

I'm trying to connect to some bluetooth devices, however I randomly get the error `Device is not connectable: 00:01:cc:94:16:0fNobleBindings.connect`. The error points to 

    	if (!deviceRecord.connectable) {
		throw new Error("Device is not connectable: " + deviceRecord.formattedAddress);
	}

https://github.com/jasongin/noble-uwp/blob/master/lib/bindings.js#L99-L101

It doesn't always happen, and sometimes I can connect to up to 10 devices without error. However, when the issue occurs it is usually after the 3rd device being connected (but I have seen it happen on the one 2nd also). It is also usually after the 2nd attempt of connecting a bunch of devices, after a clean restart, so I'm wondering if it is some caching issue - maybe with `advertisementType`?

Doing some console.log'ing, it appears that the peripheral is in a `connecting` state internally, if that helps.

Any ideas?

Thanks!

jasongin, 2017-07-24:
Can you check what is the `BluetoothLEAdvertisementType` for the problematic device? It is printed by the debug logger at https://github.com/jasongin/noble-uwp/blob/master/lib/bindings.js#L457. To enable debug output, set the `DEBUG` environment variable to `"noble-uwp"`.

taktran, 2017-07-25:
Hi,

I think the issue is an async issue, with when `_onAdvertisementWatcherReceived` is called.

From some console logging, I see that the `Device is not connectable` error occurs when `_onAdvertisementWatcherReceived` is called when the device doesn't have a name and has `advertisementType === 4`, which I'm guessing is [ScanResponse](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementtype)).

Subsequent calls to `_onAdvertisementWatcherReceived` updates the name here: https://github.com/jasongin/noble-uwp/blob/master/lib/bindings.js#L508-L510, however, `connectable` doesn't seem to be updated, hence resulting in the error when a user tries to run `.connect`.

It works for devices when the first call to `_onAdvertisementWatcherReceived` has a localName, and `advertisementType === 0`, which happens most of the time.

Not sure what the fix is here. Do you know what the usual flow of advertisement type discoveries should be? Maybe the `connectable` value needs to favour `connectable` advertisement types and update the device records accordingly?

jasongin, 2017-07-25:
I suspected it was something related to that.

I'm wondering whether [active scanning](https://github.com/jasongin/noble-uwp/blob/master/lib/bindings.js#L53) should be disabled. In that case there would be no ScanReponse advertisements received, and devices would be "discovered" after only the first advertisement. But for some devices, some of the advertisement data (including device name) is only available in an active scan response. Whatever noble does on other platforms should be matched here, I'm just not sure exactly what that is.

jasongin, 2017-07-26:
I think I understand the problem: when a "scan response" is received for a device for which an initial advertisement was not seen, the device info that is passed to the discovery event would be incomplete. That could happen when doing multiple scans in succession.

I'm currently revising the advertisement parsing logic as part of the [service UUID filtering](https://github.com/jasongin/noble-uwp/issues/23) work. I think this bug should be fixed after that's done.

jasongin, 2017-07-28:
I think I've fixed the problem in version 0.5.0. Can you confirm?

taktran, 2017-07-31:
Hi Jason, it seems to be fixed with 0.5.1.

I was able to connect to 5 devices several times with no problems.

Thanks for looking into it!

closed 2017-07-31 by taktran