jasongin/noble-uwp #27: [BUILD] wrong bindings on npm rebuild with different platform or node api ver

farfromrefug, 2017-08-01:
There is an issue with the way you load and name bindings in the modules. You always use "binding.node".

Let me explain:
- i build my app on my current node install version =>
- noble-uwp build process finds that i can remote install using your releases in github
- GREAT
- I now rebuild my app for electron which uses a different version of node (we usually do this ```npm rebuild --runtime=electron --target=1.7.5 --arch=x64 --rebuild --update-binary```)
- node-pre-gyp checks if it needs to rebuild the noble-uwp modules bindings.
- it sees that there are already ```./lib/binding/binding.node``` so it thinks it's good and tells you it does not need to rebuild

But you end up with bindings built for the wrong version of node.
The issue is that in your modules ```package.json``` in the ```binary``` section, you dont use ```node_abi```

here is a working example:
https://github.com/peterbraden/node-opencv/blob/master/package.json

jasongin, 2017-08-01:
Yes, I had noticed that problem recently. Your explanation helps me understand the issue better. I'll fix this soon. Thanks!

jasongin, 2017-08-02:
Fixed in noble-uwp 0.5.3

closed 2017-08-02 by jasongin

farfromrefug, 2017-08-02:
that was fast! Thanks