jasongin/noble-uwp #30: Error in build from source

itome, 2017-08-20:
I tried to rebuild modules by `npm run install --build-from-source` but failed with next log.
```

> noble-uwp@0.5.4 install C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp
> install.cmd

Installing NodeRT UWP adapter for Windows.Foundation
このソリューション内のプロジェクトを 1 度に 1 つずつビルドします。並行ビルドを有効にするには、"/m" スイッチを追加してく ださい。
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
  CollectionsConverterUtils.cpp
..\_nodert_generated.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly sear
ch path using /AI or by setting the LIBPATH environment variable [C:\Users\itome\Projects\MaBeeeElectron\node_modules\n
oble-uwp\uwp\windows.foundation\build\binding.vcxproj]
..\NodeRtUtils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search pat
h using /AI or by setting the LIBPATH environment variable [C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-u
wp\uwp\windows.foundation\build\binding.vcxproj]
..\CollectionsConverterUtils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assem
bly search path using /AI or by setting the LIBPATH environment variable [C:\Users\itome\Projects\MaBeeeElectron\node_m
odules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
..\OpaqueWrapper.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search p
ath using /AI or by setting the LIBPATH environment variable [C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble
-uwp\uwp\windows.foundation\build\binding.vcxproj]
  win_delay_load_hook.cc
C:\Users\itome\Projects\MaBeeeElectron\node_modules\node-gyp\src\win_delay_load_hook.cc : fatal error C1107: could not
find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment
variable [C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\itome\Projects\MaBeeeElectron\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\itome\\Projects\\MaBeeeElectron\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--msvs_version=2015" "--module=C:\\Users\\itome\\Projects\\MaBeeeElectron\\node_modules\\noble-uwp\\uwp\\windows.foundation\\binding\\node-v48-win32-x64\\binding.node" "--module_name=binding" "--module_path=C:\\Users\\itome\\Projects\\MaBeeeElectron\\node_modules\\noble-uwp\\uwp\\windows.foundation\\binding\\node-v48-win32-x64"
gyp ERR! cwd C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation
gyp ERR! node -v v6.11.2
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\itome\Projects\MaBeeeElectron\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation\binding\node-v48-win32-x64\binding.node --module_name=binding --module_path=C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation\binding\node-v48-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:891:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 10.0.15063
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\itome\\Projects\\MaBeeeElectron\\node_modules\\noble-uwp\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build" "--msvs_version=2015"
node-pre-gyp ERR! cwd C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation
node-pre-gyp ERR! node -v v6.11.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.36
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\itome\Projects\MaBeeeElectron\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation\binding\node-v48-win32-x64\binding.node --module_name=binding --module_path=C:\Users\itome\Projects\MaBeeeElectron\node_modules\noble-uwp\uwp\windows.foundation\binding\node-v48-win32-x64' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! noble-uwp@0.5.4 install: `install.cmd`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the noble-uwp@0.5.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\itome\AppData\Roaming\npm-cache\_logs\2017-08-20T03_16_02_323Z-debug.log
```

Any ideas about this?

jasongin, 2017-08-20:
Do you have Windows 10 SDK 10563 installed, as listed in the readme? Or is it installed in a non-default location?

itome, 2017-08-20:
I installed it on custom location, I reiinstall it and rebuild successfully. Thank you for your advice.

closed 2017-08-20 by itome