jasongin/noble-uwp #33: doesn't work on 32 bit systems?

Janneman84, 2017-08-25:
I am working on a 64 bit Windows, but I'm making an Electron app that is supposed to work on 32 bit systems as well. Therefore I'm making ia32 builds, that run fine on 32 and 64 systems.

Now I'm trying to make noble-uwp work here. It builds and runs with no errors. However the bluetooth state keeps being 'unsupported'. It looks like it's not discovering any bluetooth adapters.

It works fine in x64 Electron builds but not in ia32 builds. Anyone have any idea what the problem may be?

daumling, 2017-10-13:
Same here. It does not find any BT radio. I did not try an x64 Electron build, however.

Janneman84, 2017-10-13:
32 bit builds on 32 bit Windows works and 64 bit builds on 64 bit Windows works too. Only 32 bit builds on 64 bit Windows does not work. This is really unfortunate because I already have a 32 bit app deployed for over a year now. This app supports auto updates so I want my users to be able to update the app automatically with working bluetooth, regardless of it's a 32 bit of 64 bit system.

lilliesAndRoses, 2018-08-02:
Do you mean that if I have to build an app to work on 32 bit Windows, I need to **build** the app using on a 32 bit Windows machine?
I am able to build an app using arch=ia32, but it gives runtime error of bindings.node not being found.
Have there been any new developments around this that fix the problem?

kinyoklion, 2019-01-16:
The Radio.GetRadiosAsync method is used in order to get the radio state. This method is documented to only work in win32 apps on the matching architecture.

https://docs.microsoft.com/en-us/uwp/api/windows.devices.radios.radio.getradiosasync

If a radio is assumed to be on, and then scanning is started, then discovery and pairing all work regardless of system architecture.