jasongin/noble-uwp #34: notifications not working (apparently Windows issue)

Janneman84, 2017-08-25:
Notifications didn't seem to work on my and my colleague's computer using noble-uwp in an x64 Electron build. Scanning and connecting works though.

Turns out there apparently is an issue in Windows that is causing this. Regular UWP apps work fine but the problem is specific with 'external' use like dlls.

Supposedly there is a Windows Update that fixes this, but probably only people in the fast ring have this (which I'm not).

There is a topic that addresses this issue:

https://social.msdn.microsoft.com/Forums/en-US/58da3fdb-a0e1-4161-8af3-778b6839f4e1/bluetooth-bluetoothledevicefromidasync-does-not-complete-on-10015063?forum=wdk#ad57e5ff-0699-4868-ad04-b6fdf19859de

Fortunately the regedit fix that is proposed here does work for me and my colleague. However this is not really a realistic solution for my customers of course. So I'm wondering if it's possible that one of the solutions mentioned in the link can be integrated in node-uwp somehow?

jasongin, 2017-08-26:
The code I added at [`uwp/InitializeSecurity.h`](https://github.com/jasongin/noble-uwp/blob/master/uwp/InitializeSecurity.h) is specifically to address this issue. It implements the `CoInitializeSecurity()` workaround as described in the post there.

Unfortunately for a few releases of `noble-uwp`, that code was accidentally omitted from the build. But then I fixed it a few weeks ago (0d9754831dacb2e7869357d22a15c9350039d5bf). The `noble-uwp 0.5.2` release has that fix, and I verified notifications were working for me at that time, in a node command-line app.

But perhaps something is different when running in Electron? Maybe it calls `CoInitializeSecurity()` in a different way that conflicts?

geovie, 2017-09-01:
I tried it with the [sample](https://github.com/sandeepmistry/noble/wiki/Getting-started#getting-notified-on-characteristic-value-change) from noble and it does not work for me with latest noble-uwp from master (with noble and usb adapter it works).
`InitializeSecurity()` is called and there are no other errors...

@jasongin are you sure that it worked? Could you try it with the latest release?

(OS Build 15063.540, node v6.10.3)

jasongin, 2017-09-02:
Notifications do still work for me with the latest `noble-uwp`. However, I've been on more recent Windows insider builds, currently 16278. I don't have a system running Windows build 15063 available to test. (Actually with the newer Windows builds it can work even without the call to `CoInitializeSecurity()`. That explains why my testing didn't detect that regression earlier.)

I understand it would be best if things worked on non-insider builds, particularly 15063. If the `CoIinitializeSecurity()` workaround is not sufficient there, then the only other possibilities are the registry entry, or the optional windows patch that was mentioned in that forum thread. Unfortunately no KB number was given.

Janneman84, 2017-09-02:
I recommend that you install a virtual machine to test things like this out on. I prefer VMware Fusion on my Mac. It's not free but not exactly expensive.

You can download a Windows ISO here:
[https://www.microsoft.com/en-us/software-download/windows10ISO](https://www.microsoft.com/en-us/software-download/windows10ISO)
You don't need a serial key or anything, just mount the ISO and go through the installer. If you don't wish to set a profile password just keep the password field empty. It takes about 20 minutes to be desktop ready. Then install VMware tools by clicking Virtual Machine -> install VMware tools. This activates a virtual drive with an autorun installer. Installing this allows you to copy/paste files and text between the host and VM, among other things.

Once this is done install NodeJS, Visual Studio and Windows 10 SDK.

To use Bluetooth you'll want to uncheck 'share Bluetooth devices with Windows' and check the Bluetooth host controller in the USB & Bluetooth Settings so that the BT adapter will be dedicated to the VM. You might want to install some drivers for this. 

When on Mac you can try these Apple BT drivers for use in Windows:
[32 bit (x86)](http://kb.parallels.com/Attachments/kcs-6577/AppleBluetoothBroadcomInstaller.exe) |   [64 bit (x64)](http://kb.parallels.com/Attachments/kcs-6577/AppleBluetoothBroadcomInstaller64.exe)
After installing this you may have to manually assign the drivers to the device in device manager.

Doing this all takes maybe 30 minutes. Windows may not be activated but that doesn't matter.

I suggest that you try a 32 bit Windows too to see if you can get things to work there, because I don't.

Janneman84, 2017-09-02:
Also I strongly suggest that you give Electron a try yourself. It's really easy to get something up and running:

First download a release, current latest one is 1.7.5:
[electron 1.7.5 win32 x64](https://github.com/electron/electron/releases/download/v1.7.5/electron-v1.7.5-win32-x64.zip)
[electron 1.7.5 win32 x86](https://github.com/electron/electron/releases/download/v1.7.5/electron-v1.7.5-win32-ia32.zip)
[all releases](https://electron.atom.io/releases/)
The 32 bit version should work on x64 systems too. (BTW 'win32' is the name of the platform, not to be confused with architecture)

Extract the zip, open the folder 'resources' and in there create a folder named `app`. In `resources/app` create the `package.json`, `main.js` and `index.html` files. Check the [quick start](https://github.com/electron/electron/blob/v1.7.5/docs/tutorial/quick-start.md) as to what to put in these files. Then try to run `electron.exe` to see if it works.

See if you can install noble-uwp by using the suggestions from https://github.com/jasongin/noble-uwp/issues/31. The node_modules folder should appear in the `resources/app` folder as well. 

Now edit the `index.html` and add some javascript (or create a separate js file and refer to that):
`var noble = require('electron').remote.require('noble-uwp');`
`console.log(noble);`

From here on you can just create some html buttons and javascript or something to `index.hml` to test things out. Make sure `mainWindow.openDevTools();` is put in `main.js` which opens the dev console at startup. It should be there if you took the example from the quick start.

NAllred91, 2017-10-28:
@Janneman84 Did you ever find a way around this to get this working in an Electron build for an end user?

Janneman84, 2017-10-28:
There is a Windows Update that should fix the issue. It's probably included in the Fall Creators Update, but I'm not sure. If this is the case you can check the version of Windows the user is on and prompt the user that they need to update Windows. I believe it's in process or process.versions somewhere.

NAllred91, 2018-02-15:
@Janneman84 Are you still seeing this behavior?  I'm on OS 10.0.15063.909 and Electron 1.8.2  and notifications appear to be functioning fine now.

I don't remember the exact situation that caused the issue before, so I'm hesitant to start investing time in this again if this notifications issue still exists and I'm just missing something.