jasongin/noble-uwp #37: UWP Loopback IPC

btaber, 2017-09-28:
[Question] - I am new to this technology stack.  Does noble-uwp bypass the issue of UWP apps not allowing loopback IPC, either with Electron or without?  My intention is to implement a noble-uwp server app and control BLE devices through a client on the same machine, with the client implemented as a JS React app (maybe with Electron) or through a Java client.

jasongin, 2017-09-29:
I'm not sure I understand the question... a bluetooth library doesn't impact the TCP/IP loopback restriction.

btaber, 2017-09-29:
In itself, no, but I am thinking of using noble-uwp in a node/express server app and serve up a REST API for BLE connectivity into my device, and then connect a client app running as a separate java process on the same computer.  Would that violate the loopback IPC restriction?

jasongin, 2017-09-29:
My understanding is that would be blocked if the client app is a windows store app.

As a developer you can [exempt certain apps from the loopback restriction](https://msdn.microsoft.com/en-us/library/windows/apps/dn640582.aspx#configuring_the_firewall). But it's not something you can automate at installation or expect end-users to do.

btaber, 2017-09-30:
Thank you for your response.  That is my understanding as well.  I expect that implementing a React/JS client and wrapping it in Electron would also not work, because that would still require loopback IPC?