jasongin/noble-uwp #38: NPM install tries to download non-existing resource

NoponyCallsMeChicken, 2017-10-04:
Hello,

when I try installing noble-uwp via **npm** as **root** on **Windows 10**, I get a multitude of errors. They probably go back to the first one, which is a failed download of [](https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-win32-x64.tar.gz).

This makes sense, because when I enter the address in my browser, I get a 404.

If it helps, here's the output I get: [](https://gist.github.com/NoponyCallsMeChicken/60f0e5e4e1e24d277277feab4d37c80e)

Thank You!
Nopony

jasongin, 2017-10-04:
Hmm, it looks like the `node-usb` project unpublished the v1.2.0 binary for API version 57. That was not very considerate of them...

There's a v1.2.0 release of `node-usb` that still has binaries for API version 57, but it is not published to NPM. I'm not sure what to do about this.

jasongin, 2017-10-04:
As a workaround you can try `npm install --build-from-source`, if you have build tools set up.

jasongin, 2017-10-04:
Or try with a different node version (6.x)

NoponyCallsMeChicken, 2017-10-08:
It solves the problem for now. Thank You for the reply!