jasongin/noble-uwp #40: Peripheral disconnects after 30 seconds

leinelissen, 2017-10-24:
Hello!

I am currently using noble-uwp as a replacement for noble on Windows 10 Fall Creators update, and I must say I am impressed how decent it works compared to the bluetooth-hci-socket implementation. Great work!

Currently we are investigating an issue with some of our custom-built BLE devices, where after almost exactly 30 seconds the device is disconnected from the PC. This happens on every connect, but weirdly only affects some devices. We have tested those devices with Noble on MacOS High Sierra and have encountered no issues whatsoever. Our device debug logs indicate that the device is disconnected by the PC, suggesting that the issue is either related to noble-uwp or the Windows Bluetooth API. 

I have included both the logs from both our Electron app and the device. When trying to write to a characteristic, Noble will throw an error (`failed to write characteristic for device e9b98f92883e: Error: The object has been closed.`), if the write happens more than 30 seconds after connecting.

Device Logs
```
 0> Hello World! Let's go 
 0> ACC gevonden!
 0> DEBUG: Orientatie veranderd, update BLE (4) 
 0> Orientatie verandering doorgevoerd 
 0> DEBUG: BLE connection started 
 0> DEBUG: Orientation reset 
 0> DEBUG: Orientatie veranderd na reset, update BLE (4) 
 0> Orientatie verandering doorgevoerd 
 0> Bluetooth write!
 0> 1 
 0> Bluetooth FLASH write!
 0> Bluetooth write!
 0> 2 
 0> Bluetooth write!
 0> 2 
 0> DEBUG: BLE connection disconnected 
 0> Reason disconnect: 13
```

Console Log
```
14:53:28.097 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp     flags: undefined +0ms
14:53:28.097 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp     service UUID: {8BB30001-7113-2B9C-C649-C9AD5A461F3B} +1ms
14:53:28.160 C:\Users\Lei\Documents\Github\iMO-Connect\js\mdm.js:81 Connecting to device e9b98f92883e, attempt #1
14:53:28.165 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp stopScanning() +67ms
14:53:28.165 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp keepAlive(false) => 0 +1ms
14:53:28.166 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp connect(e9b98f92883e) +0ms
14:53:28.170 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp watcher stopped +4ms
14:53:28.196 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp got bluetooth device: iMO MDM (5) +27ms
14:53:28.198 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp discoverServices(e9b98f92883e, 8bb3152371132b9cc649c9ad5a461f3b) +1ms
14:53:28.886 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp e9b98f92883e services: Array(1) +688ms
14:53:28.889 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp discoverCharacteristics(e9b98f92883e, 8bb3152371132b9cc649c9ad5a461f3b, 8bb3accc71132b9cc649c9ad5a461f3b,8bb3152571132b9cc649c9ad5a461f3b,8bb3a3e171132b9cc649c9ad5a461f3b,8bb3c01071132b9cc649c9ad5a461f3b +3ms
14:53:29.568 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp e9b98f92883e 8bb3152371132b9cc649c9ad5a461f3b characteristics: Array(4) +679ms
14:53:29.570 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp write(e9b98f92883e, 8bb3152371132b9cc649c9ad5a461f3b, 8bb3152571132b9cc649c9ad5a461f3b, (data), false) +2ms
14:53:29.804 C:\Users\Lei\Documents\Github\iMO-Connect\js\mdm.js:159 Flash was sent for device e9b98f92883e
14:53:29.805 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp notify(e9b98f92883e, 8bb3152371132b9cc649c9ad5a461f3b, 8bb3accc71132b9cc649c9ad5a461f3b, true) +235ms
14:53:30.283 C:\Users\Lei\Documents\Github\iMO-Connect\node_modules\noble\node_modules\noble-uwp\node_modules\de…:123 noble-uwp keepAlive(true) => 1 +479ms
14:53:30.284 C:\Users\Lei\Documents\Github\iMO-Connect\js\mdm.js:172 Device e9b98f92883e was successfully connected
```

jasongin, 2017-10-24:
`noble-uwp` doesn't do anything that would cause the device to disconnect after 30s as far as I know. I suspect it's a Windows problem. Can you reproduce the issue using the Windows Bluetooth APIs from C#? If so, I'd suggest asking about this on the [UWP Developer Forum](https://social.msdn.microsoft.com/Forums/windowsapps/en-US/home?forum=wpdevelop) or other appropriate support channel.

qdot, 2018-04-07:
This could be due to windows power management of the bluetooth radio connected to the host. I've had devices randomly disconnect at the 30s point when power management is turned on for a radio. Insignia adapters seem to have this on by default, while other Broadcom/CSR adapters I've tried don't.