jasongin/noble-uwp #44: Is Win 10 Falls Creator Update Supported?

lumin888, 2017-12-19:
Has anyone manage to get this to work under version 1709 of Windows 10?

I have run **npm install --global --production windows-build-tools** 

Rebooted. Then **npm install node-uwp**

I get the following error snippet

> ..\OpaqueWrapper.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable

 
Is it because NodeRT doesn't yet support it?

jasongin, 2017-12-19:
As noted in the readme, you need to install Windows SDK build 10.0.15063.

It should be possible to update to a newer SDK build, but I have not tried yet.

lumin888, 2017-12-20:
Thanks Jason, I had the latest SDK but it wouldn't build with that. Installing the 15063 worked.

CharlesStover, 2018-07-22:
I managed to get this working by installing the latest SDK and just copy-pasting the `windows.winmd` file from its directory (`C:\Program Files (x86)\Windows Kits\10\UnionMetadata\10.0.17134.0`) into the one expected by this project (`C:\Program Files (x86)\Windows Kits\10\UnionMetadata\10.0.15063.0`).

As reference material for someone who wants the exact SDK version and file:
[Windows SDK archive](https://developer.microsoft.com/en-us/windows/downloads/sdk-archive)
[Windows SDK 10.0.150.63](https://go.microsoft.com/fwlink/p/?LinkId=845298)

lilliesAndRoses, 2018-07-23:
Hello,
I am getting the same error.
I do not use Windows SDK as I do not do active development. I am only a user of noble-uwp.  I have installed windows tools using npm install --global --production windows-build-tools
How do I check the version and get to the 2015 version?

lilliesAndRoses, 2018-07-24:
Looks like only the windows-build-tools are not sufficient.
Things compiled after I installed the SDK as mentioned by @CharlesStover

Thanks.

jrobeson, 2018-08-19:
@jasongin: is updating to a newer SDK worth it? I already seen tons of deprecation errors suggesting to use different windows 10 bluetooth apis (usually related to async versions). Do the newer sdks fix any issues that exist with noble-uwp?

jasongin, 2018-08-25:
I don't think updating to a newer SDK would fix any issues. My understanding is a newer SDK would just give access to newer APIs that are not declared by the older SDK. But what really matters for bugfixes is the version of Windows at runtime, which of course can be newer than the SDK that was used when building.

I'm only aware of [a few new bluetooth-related properties added in a later SDK](https://docs.microsoft.com/en-us/windows/uwp/whats-new/windows-10-build-17134-api-diff#windowsdevicesbluetooth), which are probably not relevant to `noble-uwp`.