jasongin/noble-uwp #45: issue on some windows 10 computers

farfromrefug, 2018-01-09:
On some windows 10 computers we are facing issues with noble-uwp
the computer is running windows 10 creators update but with a very simple npm project with only noble-uwp i get this:
```
PS D:\Users\aryballe\test> node .\node_modules\noble-uwp\test.js
D:\Users\aryballe\test\node_modules\noble-uwp\uwp\windows.foundation\lib\main.js:23
  throw e;
  ^

Error: Le module spécifié est introuvable.
\\?\D:\Users\aryballe\test\node_modules\noble-uwp\uwp\windows.foundation\binding\node-v57-win32-x64\binding.node
    at Object.Module._extensions..node (module.js:672:18)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (D:\Users\aryballe\test\node_modules\noble-uwp\uwp\windows.foundation\lib\main.js:19:20)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
PS D:\Users\aryballe\test>
```
The file is there just it won't load and i think it's related to something not beeing found.

[image]

In case you see something obvious... Maybe it needs some redistributable? but i have never seen the need for it

Thanks

jasongin, 2018-01-09:
Have you built the native modules yourself, or are you using the pre-built binaries?

You could try installing the Windows 10 SDK build 15063, maybe that will add some runtime dependency, though I didn't think that was needed at runtime.

Or you could try using a [dependency walker tool](https://github.com/lucasg/Dependencies) to identify the missing dependency.

farfromrefug, 2018-01-09:
@jasongin yes dependency walker was the solution!
I actually finally figured out that noble-uwp require VC C++ redistributable 2015 Update 3 !
I had one computer without any and one with the  2015 first version.
Both were not working and are now!

Even my computer (which was working) stopped working. For some reason i did not have the update 3 anymore, only the first version.

Maybe this should be put in runtime requirements?

Thanks for looking at this.

closed 2018-01-09 by farfromrefug