jasongin/noble-uwp #46: Installing NodeRT UWP adapter for Windows.Foundation node-pre-gyp ERR!

mqmq0, 2018-01-14:
Any suggestion to solve the following error?

PS D:\noble-uwp-master> npm install

> noble-uwp@0.5.4 install D:\noble-uwp-master
> install.cmd

Installing NodeRT UWP adapter for Windows.Foundation
node-pre-gyp ERR! Tried to download(404): https://github.com/jasongin/noble-uwp/releases/download/windows_foundation_0.15063.1/binding-v0.15063.1-node-v59-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for windows.foundation@0.15063.1 and node@9.4.0 (node-v59 ABI, unknown) (falling back to source compile with node-gyp)
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
D:\noble-uwp-master\uwp\windows.foundation\build\binding.vcxproj(20,3): error MSB4019: The imported project "D:\Microsoft.Cpp.Default.props" was
 not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\berec\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:12)
gyp ERR! System Windows_NT 10.0.16299
gyp ERR! command "D:\\Program Files\\nodejs\\node.exe" "C:\\Users\\berec\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--msvs_version=2015" "--module=D:\\noble-uwp-master\\uwp\\windows.foundation\\binding\\node-v59-win32-x64\\binding.node" "--module_name=binding" "--module_path=D:\\noble-uwp-master\\uwp\\windows.foundation\\binding\\node-v59-win32-x64"
gyp ERR! cwd D:\noble-uwp-master\uwp\windows.foundation
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'D:\Program Files\nodejs\node.exe C:\Users\berec\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=D:\noble-uwp-master\uwp\windows.foundation\binding\node-v59-win32-x64\binding.node --module_name=binding --module_path=D:\noble-uwp-master\uwp\windows.foundation\binding\node-v59-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\noble-uwp-master\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:943:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
node-pre-gyp ERR! System Windows_NT 10.0.16299
node-pre-gyp ERR! command "D:\\Program Files\\nodejs\\node.exe" "D:\\noble-uwp-master\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build" "--msvs_version=2015"
node-pre-gyp ERR! cwd D:\noble-uwp-master\uwp\windows.foundation
node-pre-gyp ERR! node -v v9.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok
Failed to execute 'D:\Program Files\nodejs\node.exe C:\Users\berec\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=D:\noble-uwp-master\uwp\windows.foundation\binding\node-v59-win32-x64\binding.node --module_name=binding --module_path=D:\noble-uwp-master\uwp\windows.foundation\binding\node-v59-win32-x64' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! noble-uwp@0.5.4 install: `install.cmd`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the noble-uwp@0.5.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\berec\AppData\Roaming\npm-cache\_logs\2018-01-14T19_45_47_119Z-debug.log

When i try to run something:

PS D:\noble-uwp-master> node test.js
D:\noble-uwp-master\uwp\windows.foundation\lib\main.js:23
  throw e;
  ^

Error: Cannot find module 'D:\noble-uwp-master\uwp\windows.foundation\binding\node-v59-win32-x64\binding.node'
    at Function.Module._resolveFilename (module.js:555:15)
    at Function.Module._load (module.js:482:25)
    at Module.require (module.js:604:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (D:\noble-uwp-master\uwp\windows.foundation\lib\main.js:19:20)
    at Module._compile (module.js:660:30)
    at Object.Module._extensions..js (module.js:671:10)
    at Module.load (module.js:573:32)
    at tryModuleLoad (module.js:513:12)
    at Function.Module._load (module.js:505:3)

I attach the debug log.
[2018-01-14T19_45_47_119Z-debug.log](https://github.com/jasongin/noble-uwp/files/1629739/2018-01-14T19_45_47_119Z-debug.log)

jasongin, 2018-01-16:
The easiest solution will be to use node 8.x instead of 9.x. There are no prebuilt binaries published yet for node 9.x.

Alternatively, it should be possible to compile the code against node 9.x. For that you'll need to install the Windows SDK version 15063.

jrobeson, 2018-08-19:
I think this bug should be closed, unless you plan on providing instructions for how to configure visual studio with the required sdks.    Is there even any way to configure `windows-build-tools` to install the proper windows sdks? I'm not sure.

I have to admit that I had a bit of trouble figuring out how to install the sdks properly. I still don't know how to install it without pulling in the visualstudio editor, but I punted since I didn't have time to deal with it.