jasongin/noble-uwp #47: error building for electron 1.8.2

farfromrefug, 2018-01-16:
I just tried to build for the very latest electron 1.8.2
i fails with a lot of errors like 
```
noble-uwp\uwp\windows.foundation\node-async.h(286): error C2665: 'Nan::MakeCallback': none of
 the 3 overloads could convert all the argument types (compiling source file ..\_nodert_generated.cpp) [noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
```
I think the sources need to be regenerated.
I cant remember how you do that.
Can you guide me?

Thanks

jasongin, 2018-01-16:
The script for generating the code is at `uwp/generate-uwp.cmd`. However I don't think that will help, since it will just regenerate the same code.

Probably you need to install a newer version of the `nan` package that is compatible with the latest electron build.

farfromrefug, 2018-01-16:
@jasongin you use latest for ```nan``` package. The thing is that the error is node-async which is part of the project. So not related to ```nan``` isn't it?
The only thing that change are the node-gyp iosjs headers i think.
actually i can't wrap my head around where the issue could come from.
Needs to think about this...

jasongin, 2018-01-16:
I don't know then. It seems like it might be related to some breaking change in either newer electron or newer node version, because it works with older versions.

farfromrefug, 2018-01-17:
Yes i agree i think it might be deprecated methods gone in v8.
But what i need to figure out is how to "update" files like "node-async" for those changes.
There is no new ```nan```

jasongin, 2018-01-17:
`node-async.h` comes from the [NodeRT project](https://github.com/NodeRT/NodeRT) ([node-async.h](https://github.com/NodeRT/NodeRT/blob/master/src/NodeRTLib/ProjectFiles/node-async.h)). There have not been any updates to that file in over a year, though I don't think it needs any.

I'm not very familiar with how electron consumes node, but using a newer node version doesn't seem to be the problem. I can build `noble-uwp` successfully with node version 9.4.

farfromrefug, 2018-01-18:
@jasongin Yes electron does not use your node version. It is bundled with its own.
But you have a point i can try building ```noble-uwp``` for node (and not electron) using the corresponding node version bundled with electron 1.8.2.
Or actually it might really be a v8 compatibility issue as they upgraded it
https://github.com/electron/electron/releases/tag/v1.8.1

farfromrefug, 2018-03-02:
I actually forgot about that issue. But for some reason it now works. I have no idea why ....
Sorry for that!

closed 2018-03-02 by farfromrefug