jasongin/noble-uwp #50: discoverAllServicesAndCharacteristics() doesn't always call callback.

NAllred91, 2018-02-15:
When I make calls to `peripheral.discoverAllServicesAndCharacteristics([callback])` the callback is not always called.  It seems that if I put the call to `discoverAllServicesAndCharacteristics` in a setTimeout I can get the callback to be called more frequently, but the callback is still mostly never called.

NAllred91, 2018-02-16:
This seems to just be an issue when building with an electron app.  I can't reproduce this on my desktop using Node 6.11.5

NAllred91, 2018-02-16:
Turns out, if the device _is_ paired then the callback is never called.  If the device _is not_ paired then the callback is called.

NAllred91, 2018-02-17:
It looks like it might be because initially `services` is an empty array.  This code seems to work for now, but it doesn't handle disconnects in the middle of discovering services.
```
function discoverServices(peripheral, callback) {
    peripheral.once('servicesDiscover', (services) => {
        if(services.length === 0) {
            console.warn('empty services!')
            discoverServices(peripheral, callback)
        }
        else {
            async.each(services, (service, callback) => {
                service.discoverCharacteristics([], (err, characteristics) => {
                    if(err) callback(err)
                    else {
                        service.characteristics = characteristics
                        callback()
                    }
                })
            }, (err) => {
                callback(err, services)
            })
        }
    })
}
```

jrobeson, 2018-08-19:
@NAllred91 : does noble itself handle disconnects in the middle of discovering services?