jasongin/noble-uwp #60: Noble fallback is unsupported because of os dependency

tylergets, 2018-06-22:
https://github.com/jasongin/noble-uwp/blob/master/package.json#L10

Package.json is setup to make sure the platform is windows, however the readme states:

```
On non-Windows platforms, the benavior is unchanged from noble, while on Windows the UWP bindings are used instead of noble's Bluetooth HCI bindings.
```

Would you be open to removing the os line? We have multiplatform support in our app and we want to avoid having to check the os and switch between noble and noble-uwp

jasongin, 2018-06-25:
You're right, the intention was that it could be referenced on other platforms and just redirect to regular `noble` there. So that `os` line should be removed.

jasongin, 2018-06-25:
Fixed in version `0.6.2` and published to npm.

closed 2018-06-25 by jasongin