jasongin/noble-uwp #62: Could not get localName in advertisement data?

hanhdt, 2018-08-08:
When I'm scanning devices, but get advertisement data without localName like this:

```
advertisement: { 
     localName: null,
     txPowerLevel: null,
     manufacturerData: undefined,
     serviceUuids: [ '1803', '1804' ],
     serviceData: [ {} ]
 }
```
Do I have any workaround for this?

Thanks.

jasongin, 2018-08-25:
Typically the advertisement packets that are periodically broadcast for _passive_ discovery do not include the name and some other details. But when you use `noble-uwp` to start scanning, it does an _active_ scan, and it waits for a response to the active scan request (and that response typically does include the device name) before emitting a 'discover' event.

Are you seeing this behavior for all kinds of devices, or just a particular one?

jartholomew, 2018-08-27:
Hi, I have encountered the same issue using noble-uwp in an electron app. As @hanhdt mentioned above, localName is null on peripheral discovery:

`{  
   "localName":null,
   "txPowerLevel":4,
   "manufacturerData":{  
      "type":"Buffer",
      "data":[  107, 0, 47, 12, 64, 60  ]
   },
   "serviceUuids":[  "180d", "feee", "fea5"  ],
   "serviceData":[  {  }  ]
}`

but if I write the advertisement object to the console:

`console.dir(peripheral.advertisement);`

the localName appears:

`Object { 
   "localName": "Polar H10 34BE152A",
   "txPowerLevel":4,
   "manufacturerData":{  
      "type":"Buffer",
      "data":[  107, 0, 47, 12, 64, 60  ]
   },
   "serviceUuids":[  "180d", "feee", "fea5"  ],
   "serviceData":[  {  }  ]
}`

jartholomew, 2018-08-27:
i found a workaround from this post:  

[https://stackoverflow.com/questions/17546953/cant-access-object-property-even-though-it-exists-returns-undefined](url)

```
setTimeout(function()
{
    var foo = peripheral; 

}, 100);
```

hanhdt, 2018-08-28:
Yes, it works for me. I can get localName after deplayed periods.

Thank you @jartholomew and @jasongin .

closed 2018-08-28 by hanhdt