jasongin/noble-uwp #69: advertisement discovery is really slow

jrobeson, 2018-08-23:
Using the `advertisment-discovery.js` script from examples enabling `allowDuplicates` in `startScanning` i get results between say 5 and 12 seconds for a device that advertises once every 500ms. I get much quicker responses when using the hci-socket binding.

I'm using a BCM20702A based dongle and node 10.6.0

Can anybody else replicate this sort of behavior?

jrobeson, 2018-08-23:
and DeviceIoControl isn't allowed in uwp apps?

jasongin, 2018-08-25:
`DeviceIOControl` isn't allowed in UWP apps published to the store. (Although developer / sideloaded UWP apps can call any Win32 APIs.) Even if it was allowed, it would be difficult to call from JavaScript, since the NodeRT bindings only expose WinRT APIs.

What about adjusting the `BluetoothLEAdvertisementWatcher.MinSamplingInterval` property... I don't see that discussed on that stackoverflow page, but it seems like _maybe_ it could help?

jrobeson, 2018-08-25:
what's the default sampling interval?  I tried reading the docs, but I couldn't see what the defaults for any of those properties in  `BluetoothLEAdvertisementWatcher`

jrobeson, 2018-08-25:
Can you tell me about what kind of numbers you see? and what settings your peripherals have?

jasongin, 2018-08-25:
I'm actually on vacation now, I won't be able to really look into this for at least a couple weeks.

tony-gutierrez, 2021-04-01:
Did anyone get a way to get better scan timings in UWP?