jasongin/noble-uwp #71: Can not compile with electron

fedyfausto, 2018-11-22:
When i try to build for electron (npm rebuild --runtime=electron --target=3.0.8 --arch=x64 --rebuild --disturl=https://atom.io/download/electron --build_from_source=true) i get this error:

> noble-uwp@0.6.2 install D:\Progetti\Domo\ui\node_modules\noble-uwp
> node install.js

Installing NodeRT UWP adapter for Windows.Foundation
Compilazione dei progetti nella soluzione uno alla volta. Per abilitare la compilazione parallela, aggiungere l'opzione "/m".
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
  CollectionsConverterUtils.cpp
  win_delay_load_hook.cc
..\OpaqueWrapper.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
..\_nodert_generated.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
..\CollectionsConverterUtils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
C:\Users\fedyf\AppData\Roaming\nvm\v8.11.2\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.cc : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
..\NodeRtUtils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\fedyf\AppData\Roaming\nvm\v8.11.2\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\fedyf\\AppData\\Roaming\\nvm\\v8.11.2\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--msvs_version=2015" "--module=D:\\Progetti\\Domo\\ui\\node_modules\\noble-uwp\\uwp\\windows.foundation\\binding\\electron-v3.0-win32-x64\\binding.node" "--module_name=binding" "--module_path=D:\\Progetti\\Domo\\ui\\node_modules\\noble-uwp\\uwp\\windows.foundation\\binding\\electron-v3.0-win32-x64"
gyp ERR! cwd D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation
gyp ERR! node -v v8.11.2
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\fedyf\AppData\Roaming\nvm\v8.11.2\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\binding\electron-v3.0-win32-x64\binding.node --module_name=binding --module_path=D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\binding\electron-v3.0-win32-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\Progetti\Domo\ui\node_modules\noble-uwp\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:126:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:925:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
node-pre-gyp ERR! System Windows_NT 10.0.17134
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\Progetti\\Domo\\ui\\node_modules\\noble-uwp\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build" "--msvs_version=2015"
node-pre-gyp ERR! cwd D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation
node-pre-gyp ERR! node -v v8.11.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.36
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\fedyf\AppData\Roaming\nvm\v8.11.2\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\binding\electron-v3.0-win32-x64\binding.node --module_name=binding --module_path=D:\Progetti\Domo\ui\node_modules\noble-uwp\uwp\windows.foundation\binding\electron-v3.0-win32-x64' (1)

itome, 2019-03-05:
@fedyfausto You may need to install Window 10 SDK 10.0.15063.468 from following link.
https://developer.microsoft.com/ja-jp/windows/downloads/sdk-archive

acstacey, 2019-03-08:
@fedyfausto I encountered that error after installing the correct SDKs but didn't have visual studio 2015.  I believe it's possible to avoid installing 2015 by changing paths and/or copying the 2017 windows.winmd to the expected place but I solved it by doing a minimal install of visual studio 2015.

If you're not making your own changes, I have done electron builds and released them on my fork.