jasongin/noble-uwp #78: Electron app crashes due to noble-uwp

pambardekar, 2019-04-15:
I am using electron app to connect to multiple devices. The app crashes randomly (not every time) after different amount of up time. I was able to capture stack trace during one of the crashes. I have attached it.
[image]

jasongin, 2019-04-15:
Looks like a bug in [NodeRT](https://github.com/NodeRT/NodeRT). Unfortunately that project isn't very actively maintained anymore. Another reason to [consider not using NodeRT for this project](https://github.com/jasongin/noble-uwp/issues/4). But I personally don't have time to work on a replacement.

sonamgupta-kiwi, 2020-11-15:
Any update !! getting the same crash !! How to resolve this?