jasongin/noble-uwp #82: Need "step by step" install instructions

dsteinman, 2020-04-23:
Hi, I'm trying to get my code for noble working in Windows.

I have had a lot of trouble trying to install noble-uwp.  Lots of node-gyp error, missing "windows.winmd" and "platform.winmd" error.

I've also tried these 2 forks:

- https://github.com/acstacey/noble-uwp
- https://github.com/tylerc/noble-uwp/tree/fix-nodejs-12.x

I have installed multiple versions of Visual Studio, both 2015, and 2017, the Windows 10 SDK.  Which combinations of these things are actually required?

The problem that's giving me trouble at the moment is the location of **'Windows.winmd'**

```c:\users\dan\dev\uwp\noble-uwp\uwp\windows.foundation\collectionsconverterutils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]```

I've tried copying it to `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\lib\x86\store\references` but it didn't work.  That error message is not helpful without knowing where it is actually looking for the file.

Here's my log after a `git clone` and `npm install`

```
$ npm install

> usb@1.6.3 install C:\Users\Dan\dev\uwp\noble-uwp\node_modules\usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.3
prebuild-install info looking for cached prebuild @ C:\Users\Dan\AppData\Roaming\npm-cache\_prebuilds\917161-usb-v1.6.3-node-v79-win32-x64.tar.gz
prebuild-install info found cached prebuild
prebuild-install info unpacking @ C:\Users\Dan\AppData\Roaming\npm-cache\_prebuilds\917161-usb-v1.6.3-node-v79-win32-x64.tar.gz
prebuild-install info unpack resolved to C:\Users\Dan\dev\uwp\noble-uwp\node_modules\usb\build\Release\usb_bindings.node
prebuild-install info unpack required C:\Users\Dan\dev\uwp\noble-uwp\node_modules\usb\build\Release\usb_bindings.node successfully
prebuild-install info install Successfully installed prebuilt binary!

> bluetooth-hci-socket@0.5.1 install C:\Users\Dan\dev\uwp\noble-uwp\node_modules\bluetooth-hci-socket
> node-gyp rebuild

C:\Users\Dan\dev\uwp\noble-uwp\node_modules\bluetooth-hci-socket>if not defined npm_config_node_gyp (node "C:\Users\Dan\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\Dan\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  win_delay_load_hook.cc
  binding.vcxproj -> C:\Users\Dan\dev\uwp\noble-uwp\node_modules\bluetooth-hci-socket\build\Release\\binding.node

> noble-uwp@0.6.2 install C:\Users\Dan\dev\uwp\noble-uwp
> node install.js

Installing NodeRT UWP adapter for Windows.Foundation
node-pre-gyp WARN Using needle for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/jasongin/noble-uwp/releases/download/windows_foundation_0.15063.1/binding-v0.15063.1-node-v79-win32-x64.tar.gz
node-pre-gyp WARN Pre-built binaries not found for windows.foundation@0.15063.1 and node@13.13.0 (node-v79 ABI, unknown) (falling back to source compile with node-gyp)
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
  CollectionsConverterUtils.cpp
  win_delay_load_hook.cc
c:\users\dan\dev\uwp\noble-uwp\uwp\windows.foundation\_nodert_generated.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\appdata\roaming\npm\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.cc : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\uwp\noble-uwp\uwp\windows.foundation\nodertutils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\uwp\noble-uwp\uwp\windows.foundation\collectionsconverterutils.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\uwp\noble-uwp\uwp\windows.foundation\opaquewrapper.cpp : fatal error C1107: could not find assembly 'Windows.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable [C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Dan\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Windows_NT 10.0.18363
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Dan\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--msvs_version=2015" "--module=C:\\Users\\Dan\\dev\\uwp\\noble-uwp\\uwp\\windows.foundation\\binding\\node-v79-win32-x64\\binding.node" "--module_name=binding" "--module_path=C:\\Users\\Dan\\dev\\uwp\\noble-uwp\\uwp\\windows.foundation\\binding\\node-v79-win32-x64" "--napi_version=6" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=node-v79"
gyp ERR! cwd C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation
gyp ERR! node -v v13.13.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Dan\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\binding\node-v79-win32-x64\binding.node --module_name=binding --module_path=C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\binding\node-v79-win32-x64 --napi_version=6 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v79' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\Dan\dev\uwp\noble-uwp\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:1026:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
node-pre-gyp ERR! System Windows_NT 10.0.18363
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Dan\\dev\\uwp\\noble-uwp\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build" "--msvs_version=2015"
node-pre-gyp ERR! cwd C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation
node-pre-gyp ERR! node -v v13.13.0
node-pre-gyp ERR! node-pre-gyp -v v0.14.0
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Dan\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --msvs_version=2015 --module=C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\binding\node-v79-win32-x64\binding.node --module_name=binding --module_path=C:\Users\Dan\dev\uwp\noble-uwp\uwp\windows.foundation\binding\node-v79-win32-x64 --napi_version=6 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v79' (1)
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules\noble\node_modules\xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

added 102 packages from 56 contributors and audited 209 packages in 10.896s

2 packages are looking for funding
  run `npm fund` for details

found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details

```

I would really appreciate any help, or maybe even suggestions like any other libraries, forks, or projects to get BLE working in Node on Windows?

jasongin, 2020-04-23:
Did you install Windows 10 SDK build 10.0.15063 as listed in the requirements at the top of the readme? You need that specific version, not any later version (unless you update the referenced version in `common.gypi`). Get it from [here](https://developer.microsoft.com/en-us/windows/downloads/sdk-archive/).

The search paths for the `.winmd` files are the directories listed [here](https://github.com/jasongin/noble-uwp/blob/a76c6230720f768615faccf5b5dfb41cc0b8ed33/common.gypi#L30-L33).

dsteinman, 2020-04-23:
Thanks @jasongin, installing that version did resolve the missing files, and then, as I saw in other threads, I encountered a slew of node RT errors.

Any chance that'll ever be fixed?

```
Installing NodeRT UWP adapter for Windows.Foundation
node-pre-gyp WARN Using needle for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/jasongin/noble-uwp/releases/download/windows_foundation_0.15063.1/binding-v0.15063.1-node-v79-win32-x64.tar.gz
node-pre-gyp WARN Pre-built binaries not found for windows.foundation@0.15063.1 and node@13.13.0 (node-v79 ABI, unknown) (falling back to source compile with node-gyp)
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  _nodert_generated.cpp
  NodeRtUtils.cpp
  OpaqueWrapper.cpp
  CollectionsConverterUtils.cpp
  win_delay_load_hook.cc
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(34): error C2039: 'Handle': is not a member of 'v8' (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
  c:\users\dan\appdata\local\node-gyp\cache\13.13.0\include\node\v8-platform.h(16): note: see declaration of 'v8' (compiling source file ..\NodeRtUtils.cpp)
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(34): error C2061: syntax error: identifier 'Handle' (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(36): error C2065: 'value': undeclared identifier (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): error C2039: 'Handle': is not a member of 'v8' (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
  c:\users\dan\appdata\local\node-gyp\cache\13.13.0\include\node\v8-platform.h(16): note: see declaration of 'v8' (compiling source file ..\NodeRtUtils.cpp)
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): error C2065: 'Handle': undeclared identifier (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): error C2275: 'v8::Value': illegal use of this type as an expression (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
  c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): note: see declaration of 'v8::Value' (compiling source file ..\NodeRtUtils.cpp)
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): error C2065: 'hiddenVal': undeclared identifier (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): error C2065: 'value': undeclared identifier (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(41): error C2059: syntax error: ')' (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(42): error C2065: 'hiddenVal': undeclared identifier (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\opaquewrapper.h(47): error C2065: 'hiddenVal': undeclared identifier (compiling source file ..\NodeRtUtils.cpp) [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\nodertutils.cpp(22): error C2039: 'Handle': is not a member of 'v8' [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
  c:\users\dan\appdata\local\node-gyp\cache\13.13.0\include\node\v8-platform.h(16): note: see declaration of 'v8'
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\nodertutils.cpp(22): error C2873: 'Handle': symbol cannot be used in a using-declaration [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\nodertutils.cpp(110): warning C4996: 'Nan::MakeCallback': was declared deprecated [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
  c:\users\dan\dev\noble-uwp\node_modules\nan\nan.h(1047): note: see declaration of 'Nan::MakeCallback'
c:\users\dan\dev\noble-uwp\uwp\windows.foundation\nodertutils.cpp(146): error C2664: 'v8::String::Value::Value(const v8::String::Value &)': cannot convert argument 1 from 'v8::Local<v8::Value>' to 'const v8::String::Value &' [C:\Users\Dan\dev\noble-uwp\uwp\windows.foundation\build\binding.vcxproj]
  c:\users\dan\dev\noble-uwp\uwp\windows.foundation\nodertutils.cpp(146): note: Reason: cannot convert from 'v8::Local<v8::Value>' to 'const v8::String::Value'
  c:\users\dan\dev\noble-uwp\uwp\windows.foundation\nodertutils.cpp(146): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
[... cut ...] 0 --node_napi_label=node-v79' (1)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules\xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

audited 209 packages in 7.343s

2 packages are looking for funding
  run `npm fund` for details

found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
```

jasongin, 2020-04-23:
A fix for those build errors is in the fork: https://github.com/jasongin/noble-uwp/compare/master...tylerc:fix-nodejs-12.x
PR is here: https://github.com/jasongin/noble-uwp/pull/80

Sorry this project isn't in a great state now, and I haven't had time or motivation to maintain it.

dsteinman, 2020-04-23:
Thanks @jasongin I'll give it a try, and I appreciate the help.

closed 2020-04-23 by dsteinman