jasongin/noble-uwp #85: discoverSomeServices uses discoverAll (slow), and getGattServicesForUuidAsync does not accept valid guids

pursual, 2020-09-02:
When telling this binding to discover a specific list of service uuids, it instead discovers all, and then filters the results. This is slow if you are working with a device with many services. 

I attempted to fix this, but no matter what I throw at getGattServicesForUuidAsync, it always returns "Bad arguments: No suitable overload found".

pursual, 2020-09-24:
It doesnt appear that any methods in bindings JS call any of the C methods that accept a guid as an argument, so it is entirely possible that passing a guid to a C method is just broken.

tony-gutierrez, 2020-09-24:
Figured out what was going on with the uuids. See https://github.com/jasongin/noble-uwp/pull/86

pursual, 2020-09-24:
Closing for now as there is a path forward, or users can use the code in the PR

closed 2020-09-24 by pursual

jasongin, 2020-09-24:
I merged the PR. Thanks @tony-gutierrez !