jasongin/noble-uwp #88: Unable to install on windows 10 with any version of node

Deca, 2020-10-15:
Hi,
I'm trying to install the module on windows 10 but no matter what npm version I'm using (10.x,11.x,12.x.14.x) I get the same error below.
I've also run npm install --global --production windows-build-tools and installed both the 2019 and 2017 version of Visual Studio Community. Same problem. 

```
C:\Users\User\Documents\www\noble-uwp>npm i

> noble-uwp@0.6.2 install C:\Users\User\Documents\www\noble-uwp
> node install.js

Installing NodeRT UWP adapter for Windows.Foundation
node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/jasongin/noble-uwp/releases/download/windows_foundation_0.15063.1/binding-v0.15063.1-node-v72-win32-x64.tar.gz
node-pre-gyp WARN Pre-built binaries not found for windows.foundation@0.15063.1 and node@12.16.0 (node-v72 ABI, unknown) (falling back to source compile with node-gyp)
gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
gyp ERR! find VS - looking for Visual Studio version 2015
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS checking VS2017 (15.9.28307.1274) found at:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community"
gyp ERR! find VS - found "Visual Studio C++ core features"
gyp ERR! find VS - found VC++ toolset: v141
gyp ERR! find VS - found Windows SDK: 10.0.17763.0
gyp ERR! find VS - msvs_version does not match this version
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - found in "C:\Program Files (x86)\Microsoft Visual Studio 14.0"
gyp ERR! find VS - could not find MSBuild in registry for this version
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS valid versions for msvs_version:
gyp ERR! find VS - "2017"
gyp ERR! find VS - "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community"
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
gyp ERR! configure error
gyp ERR! stack Error: Could not find any Visual Studio installation to use
```

finaldzn, 2023-01-11:
Hello, i know this is 2 years old, but have you managed to install it ? im running into install error after install error...

Deca, 2023-01-11:
Unfortunately no, I had to abandon the idea of using noble under windows

finaldzn, 2023-01-12:
So I ended up suceeding for any one strugglign and coming into this issue : 

- first use this fork https://github.com/tylerc/noble-uwp/tree/fix-nodejs-12.x
- downgrade node to 12.x
- Install visual studio community <= 2019
- Install the visual c++ build tools
- Install the correct windows SDK kit (10.0.15063)
- I had a problem with some files not recognized so i had to do some shanenigans that i followed [here](https://stackoverflow.com/questions/42157554/nodert-could-not-find-assembly)
- then run `npm install --build-from-source` (if you don't see node gyp, you might need to run `npm i` inside the node_modules/noble-uwp)
- you might have to install windows SDK kit (10.0.19041.0) for @abandoware/noble

I was using @abandoware/noble so some api where not working with the windows binding (like noble.reset)

After that it was working !